m5stack/M5StickC-Plus #1: AXP192.cpp SetSleep() is different than the one for M5StickC

patjazz, 2020-09-24:
M5.Axp.DeepSleep() is OK with the M5StickC version of SetSleep(), but not with the one supplied for M5StickC-Plus.... is this a bug ?

the correction I made in AXP192.cpp:

void AXP192::SetSleep(void) //***********version M5stick-C**********
{
    Write1Byte(0x31 , Read8bit(0x31) | ( 1 << 3)); // Turn on short press to wake up
    Write1Byte(0x90 , Read8bit(0x90) | 0x07); // GPIO1 floating
    Write1Byte(0x82, 0x00); // Disable ADCs
    Write1Byte(0x12, Read8bit(0x12) & 0xA1); // Disable all outputs but DCDC1
}
/*
void AXP192::xSetSleep(void) //*********version M5stick-C plus*********
{
    uint8_t buf = Read8bit(0x31);
    buf = (1<<3)|buf;
	Write1Byte( 0x31 , buf );
	Write1Byte( 0x90 , 0x00 );
	Write1Byte( 0x12 , 0x09 );
    Write1Byte( 0x12 , 0x00 );
}
*/

DanieleQ97, 2020-09-27:
Yeah, there's something wrong with it: I was trying to wake the ESP with the motion-detection interrupt from the IMU but SetSleep prevented that. The old stickc version of the function instead works.

Bit-shift and bitwise operators are a bit too "low-level" for my poor little brain to understand fully what's going on here but, if I had to guess, I would say some hardware changes (maybe adding the buzzer) also changed the addresses of on-board perifericals and forced them to update the function that now turns off something it shouldn't.

RocketManRC, 2020-09-30:
The code from the original M5stick-C appears to work at first glance but it is leaving the processor running. On the other hand the M5stick-C plus code turns everything off but when the power switch is pressed it starts up waiting to download instead of a hard reset to startup properly. Using the datasheet for the AXP192 and Google Translate I discovered that the outputs during sleep had been changed to open collector (from floating) but changing it back to floating didn't work either. After some experimenting I found that the "low noise" setting seems to work reliably. 

Here is my local function that does the job:

void SetSleep(void) //version M5stick-C plus
{
  Write1Byte(0x31, Read8bit(0x31) | ( 1 << 3)); // short press to wake up
  Write1Byte(0x90, Read8bit(0x90) | 0x02); // GPIO1 "low noise" otherwise it goes to download mode when waking up
  Write1Byte( 0x12 , 0x00 );  // all outputs off
}

and local read and write functions to go with it:

uint8_t Read8bit( uint8_t Addr )
{
	Wire1.beginTransmission(0x34);
	Wire1.write(Addr);
	Wire1.endTransmission();
	Wire1.requestFrom(0x34, 1);
	return Wire1.read();
}

void Write1Byte( uint8_t Addr ,  uint8_t Data )
{
	Wire1.beginTransmission(0x34);
	Wire1.write(Addr);
	Wire1.write(Data);
	Wire1.endTransmission();
}

The datasheet is here: http://www.x-powers.com/en.php/Info/down/id/50

pguyot, 2020-10-24:
M5StickCPlus SetSleep version obviously doesn't work. The fact that it writes register 12H twice is suspect.
AXP192 drive code from M5StickC seems more recent and is better documented, and this version of SetSleep function actually works as @patjazz found.

From what I understand, writing 0x02 to register 0x90 effectively sets pin 19 to LDOio0 (instead of GPIO0), which is used for microphone on both M5StickC and M5StickCPlus. It is done in AXP192::begin() function of both drivers. Writing 0x07 to this register is done in the M5StickC driver elsewhere to effectively disable LD0. @RocketManRC where did you read anything related to setting the device in download mode?

RocketManRC, 2020-10-26:
@pguyot What I discovered by experiment is that with the original code for the M5StickC-Plus the processor wakes up in download mode when the power switch is pushed. After that it requires a power cycle to startup properly.

felmue, 2020-11-14:
Hi guys

I concur in that the current `SetSleep()` function is not working as expected in regard of using deep sleep (or light sleep) functionality of ESP32. As pointed out for deep sleep (or light sleep) to work, the ESP32 chip needs to stay powered on which can be achieved, as pointed out by @RocketManRC, by keeping DCDC1 on.

What the fine engineers at M5Stack have done differently hardware wise from M5StickC to M5StickC Plus is that they connected the RTC interrupt output to the AXP192 power key. That allows to completely shutdown ESP32 (except RT) and have it turned back on via RTC timer or alarm. (On M5StackC the RTC interrupt output is connected to GPIO35.)

I wonder if within the M5Stack team there is a confusion between ESP32s deep sleep / light sleep functionality and something they also call deep sleep but actually means to shut down everything except RTC.

In my opinion the AXP SetSleep() function should be restored to something similar as used for M5StickC and do the necessary preparation work for ESP32 deep sleep / light sleep so that AXP DeepSleep() and LightSleep() functions can work properly.

And for the case where the system needs to be fully shut down (except RTC) the AXP PowerOff() function should be reinstated.

BTW: The same confusion exists for M5Core2.

Thanks
Felix

felmue, 2020-11-14:
Hi guys

Re download mode: The reason, I think, that the setting for AXP GPIO0 matters when ESP32 is powered on is the fact that AXP GPIO0 (when set as LDO) powers the microphone. And the CLK input of the microphone is connected to GPIO0 of the ESP32 and ESP32 GPIO0 is a strapping pin which is used to determine whether to boot normally (ESP GPIO0 high) or into download mode (ESP32 GPIO0 low). It looks like when the microphone is not powered that its CLK input pulls down the line enough for it to register as low and hence forces the ESP32 into download mode.

Thanks
Felix

lilalukas, 2020-11-23:
By using the following commands, I get **2.09mA** with Grove enabled:
```
Write1Byte(0x31 , Read8bit(0x31) | ( 1 << 3)); // Turn on short press to wake up
Write1Byte(0x90 , Read8bit(0x90) | 0x07); // GPIO1 floating
Write1Byte(0x82, 0x00); // Disable ADCs
Write1Byte(0x12, Read8bit(0x12) & ~(1<<1) & ~(1<<2) & ~(1<<3)); // Disable all outputs but DCDC1 (DCDC3 = NC, LDO2 = LCD-BL, LDO3 = LCD-Logic)
```
With Grove disabled (so the AXP disables the 5V step up converter for the Grove-VCC) I get **1.96mA**

```
//Enables/Disables the 5V Boost for Grove
void AXP192::setGrovePower(bool active){
	if(active){
		Write1Byte(0x10, Read8bit(0x10) | (1<<2));
	}else{
		Write1Byte(0x10, Read8bit(0x10) &~(1<<2));
	}
}
```

pguyot, 2021-07-03:
> Update: I don't think below line is doing what the comment is implicating. According to the datasheet register 0x31 deals with power off voltage and (1 << 3) which equals to bit 3 is reserved in that register. I think below line actually does nothing.
> 
> `Write1Byte(0x31, Read8bit(0x31) | ( 1 << 3)); // short press to wake up`

In `Datasheet v1.13`, bit 3 is actually documented. A translation of this datasheet is available here:
https://github.com/rogerdahl/m5stickc-idf/blob/master/docs/AXP192%20Datasheet%20v1.13_cn.zh-CN.en.pdf

hpsaturn, 2022-02-24:
This workaround works for me:

> Write1Byte(0x12, Read8bit(0x12) & 0xA1); // Disable all outputs but DCDC1

But maybe in the new unified library from M5Stack they fixed the issue? Maybe not for this:

https://github.com/m5stack/M5Unified/blob/master/src/utility/Power_Class.cpp#L369-L377

But the workaround works. Maybe @RocketManRC should be do a pull request to them in this library?

closed 2022-07-15 by Tinyu-Zhao