m5stack/M5StickC-Plus #11: Why is this a separate library from M5StickC?

Gearme, 2020-12-30:
Given the similarities between the M5StickC and the M5StickC-Plus, it seems unnecessary to make this a completely separate library. There is already unnecessary divergence between the two libraries, e.g. the wake on motion functionality (present on M5StickC, missing here). With the platformio support already present, could we not just use build flags and platform settings to differentiate and otherwise use the same library for both?

1technophile, 2020-12-30:
It makes sense. Going further you could also propose a single library for all the M5 boards.

Lukanite, 2021-04-04:
Agreed, looking for the ability to create a project with this board, and having the wake on motion capabilities; until then I'm stuck with just the older, non-plus units

anuradhawick, 2021-04-19:
@Lukanite You mean this function `void enableWakeOnMotion(Ascale ascale, uint8_t thresh_num_lsb);` ? I am also waiting for things to be updated! Did you find any workaround?

anuradhawick, 2021-04-19:
Thanks, that's seems to be a decent workaround. I did not know AXP class was changed too. Cheers!

Tinyu-Zhao, 2023-07-21:
@anuradhawick @Lukanite @Gearme Add wake on motion function.
https://github.com/m5stack/M5StickC-Plus/commit/3ece12a8ae05c6d90756a9dec07134dbcbf9f075

If you want to use a unified API on almost all M5Stack host devices, we recommend using the [M5Unified](https://github.com/m5stack/M5Unified) library for development.

closed 2023-07-21 by Tinyu-Zhao