m5stack/M5StickC-Plus #13: Sleep causing unresponsive device

pcrowlz, 2021-01-30:
Going into LightSleep or DeepSleep causes the device to freeze or lock up. Uploading the same code by using #include <M5StickC.h> instead of #include <M5StickCPlus.h> fixes the issue, but causes other modules like the display to not function correctly.

This is what I get out of the debug:
018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124

pcrowlz, 2021-01-30:
I was able to get around it in the meantime by copying the M5StickC library (which does upload and sleep correctly, but doesn't have larger display and speaker functionality) and importing the M5StickCPlus eSPI, Speaker, and ST7789 .h and .cpp files. Some of the header files needed to be adjusted as to not define duplicate libraries.

felmue, 2021-02-02:
Hello @pcrowlz 

the issue is in `SetSleep()` which completely powers off everything (except for RTC) instead of keeping ESP32 powered so it can wake up from deep sleep (or light sleep).

(To be honest I never really understood the role of the `SetSleep()` function or why it even exists.)

Anyhow, using the functions from M5StickC should help, as you've already figured out yourself.

For reference:
M5StickC Plus - SetSleep()
```
Write1Byte( 0x12 , 0x00 );
```
M5StickC - SetSleep()
```
Write1Byte(0x12, Read8bit(0x12) & 0xA1); // Disable all outputs but DCDC1
```
Thanks
Felix

rlsharp, 2022-01-11:
Here it is 2022 and they still have not fixed the M5stickCPlus library for deep and light sleep. Seems such critical features should have been fixed a long time ago. Makes me question support or future of this product. Anyway, I fixed it by simply moving the SetSleep(), DeepSleep(), and LightSleep() routines in AXP192.cpp from the m5stickC library to the m4stickcplus library. So far so good.  Ron

Knight13th, 2022-02-08:
Thank you so much. I agree, how have they not fixed such a significant feature. I would recommend their devices, if it wasn't for the fact that I keep running into major software errors.

Tinyu-Zhao, 2022-07-15:
This issue has been fixed in the new version #31

closed 2022-07-15 by Tinyu-Zhao