m5stack/M5StickC-Plus #16: Flashing with other tools than Arduino IDE results display fail

drax2gma, 2021-03-22:
When flashing with the following tools, bad things happen (display black or garbage)
- EasyLoader_M5StickC_FactoryTest.exe
- M5Burner (any rom flashed from StickCPlus)

When using Arduino IDE, I can flash any Example without problems. 
Tried with different baud rates for upload, different partitions with Arduino IDE, always the same, Arduino IDE works, the other tools failing.

Pease advice. I want to flash at least the latest UIFlow rom.

felmue, 2021-03-24:
Hello @drax2gma 

M5StickC and M5StickC **Plus** are two different devices. The **Plus** version has a bigger display. Please try the EasyLoader **Plus** version, e.g.

https://m5stack.oss-cn-shenzhen.aliyuncs.com/EasyLoader/Windows/CORE/EasyLoader_M5StickC_Plus_FactoryTest.exe

Thanks
Felix

drax2gma, 2021-03-25:
Hello @felmue 

You were absolutely right, I've used the wrong firmware :(  Interestingly that was the one what seemed to work.
But the real problem was the 3rd party "long" cable. While I used the factory provided short one, everything went fine.
Thanks for your help!

Mark

felmue, 2021-03-25:
Hello @drax2gma 

I am glad to hear you got it working. Thanks for reporting back.

Cheers
Felix

closed 2022-06-19 by Tinyu-Zhao