m5stack/M5StickC-Plus #26: M5StickC-Plus with MLX90640

famtory, 2021-11-13:
Hello
I was using M5stickC and MLX90640. After some time, I bought the M5stickc plus. The advantage of M5stickc plus is that it has a large LCD, but when used with the MLX90640, the screen is displayed small. I am not good at Arduino, so I modified the source, but I did not get the desired result. Please modify it so that it is displayed on the full screen of m5stickc plus.

felmue, 2021-11-16:
Hello @famtory 

have you tried to use MLX90640 example for M5StickCPlus? You can find it [here](https://github.com/m5stack/M5StickC-Plus/tree/master/examples/Unit/THERMAL_MLX90640).

Thanks
Felix

felmue, 2021-11-16:
Hello @famtory 

ok, I see. I guess that is something M5Stack engineers need to fix then.

Thanks
Felix

famtory, 2022-02-21:
About 3 weeks ago, version 0.0.6 was released.
But the result is the same.
I've been waiting for 3 months...
Why don't fix it yet?

Tinyu-Zhao, 2022-03-12:
I apologize for not being able to reply in time. You can try [changing lines 22,3](https://github.com/m5stack/M5StickC-Plus/blob/9b1a17f8f51ddab20240c1857267434ad7cd1033/examples/Unit/THERMAL_MLX90640/THERMAL_MLX90640.ino#L22-L23) of the sample program to 
`
#define COLS              36
#define ROWS              30 
`

closed 2022-07-15 by Tinyu-Zhao