m5stack/M5StickC-Plus #34: Esp-idf arduino component require

rafapaezbas, 2022-03-27:
https://github.com/m5stack/M5StickC-Plus/blob/3e9db294bac4cb99cdcf5353432df936619ad7b8/CMakeLists.txt#L13

Hello m5stack team,

this is more a question than an issue. Im trying to use M5StickC-Plus as a esp-idf component. Seems like it requires another component named arduino. What is the repo for the __arduino__ component?. I can only find this -> https://github.com/espressif/arduino-esp32

Is arduino-esp32 component compatible with M5SickC-Plus?

Thank you very much and kind regards,
Rafa

alexeo, 2022-04-26:
I think you are right。

alexeo, 2022-04-26:
https://www.youtube.com/watch?v=05GKbdcH92Y 可以参考这个

ameerhazo, 2022-10-12:
Hello @rafapaezbas, are using PlatformIO to build this library?

Tinyu-Zhao, 2023-07-21:
Recommended to use M5Unified, which is currently the main maintenance of both Arduino and ESP-IDF support library.

closed 2023-07-21 by Tinyu-Zhao