m5stack/M5StickC-Plus #36: Get battery level execution issue

hpsaturn, 2022-06-24:
## Summary

Hi, with the last version of Arduino Framework and Espressif SDK, also with previous SDK, we have the next crash:

```cpp
:M5STICKCPLUS$ esp32decode -e firmware.elf 0x40083a91:0x3ffcdef00x40095ae1:0x3ffcdf10 0x4009b169:0x3ffcdf30 0x4009657e:0x3ffce060 0x401e15ba:0x3ffce0a0 0x401e171d:0x3ffce0c0 0x400de2ba:0x3ffce0e0 0x400de2f1:0x3ffce100 0x400de30c:0x3ffce130 0x400e0851:0x3ffce150 0x400d7ee7:0x3ffce170 0x400f0eba:0x3ffce1e0

/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:821 (discriminator 2)
/home/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp:339
/home/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp:503
/home/pio/canairio_firmware/.pio/libdeps/M5STICKCPLUS/M5StickCPlus/src/AXP192.cpp:121
/home/pio/canairio_firmware/.pio/libdeps/M5STICKCPLUS/M5StickCPlus/src/AXP192.cpp:66
/home/pio/canairio_firmware/.pio/libdeps/M5STICKCPLUS/M5StickCPlus/src/AXP192.cpp:327
/home/pio/canairio_firmware/lib/batterylib/battery_m5stack.cpp:14
/home/pio/canairio_firmware/src/main.cpp:233
/home/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42
```

My code is the next:
```cpp
void Battery_M5STACK::update() {
    curv = M5.Axp.GetBatVoltage();    // <== crash
    vusb = M5.Axp.GetVBusVoltage();
}
```

The wire init in my development is:
```cpp
void Sensors::startI2C() {
#if defined(M5STICKCPLUS) || defined(M5COREINK) 
    Wire.begin(32,33);   // M5CoreInk Ext port (default for all sensors)
    enableWire1();
#endif
#ifdef M5ATOM
    enableWire1();
#endif
#if not defined(M5STICKCPLUS) && not defined(M5COREINK) && not defined(M5ATOM)
    Wire.begin();
#endif
}

void Sensors::enableWire1() {
#ifdef M5STICKCPLUS
    Wire1.flush();
    Wire1.begin(0,26);   // M5CoreInk hat pines (header on top)
#endif
#ifdef M5COREINK
    Wire1.flush();
    Wire1.begin(25,26);   // M5CoreInk hat pines (header on top)
#endif
#ifdef M5ATOM
    Wire1.flush();
    Wire1.begin(26,32,100000);   // M5CoreInk Ext port (default for all sensors)
#endif
}
```

Setup:
```python
Resolving M5STICKCPLUS environment packages...
Platform espressif32 @ 4.4.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20003.0 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
├── toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch3)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3)
```

Source code:

[CanAirIO Firmware](https://github.com/kike-canaries/canairio_firmware/tree/devel)
[CanAirIO Sensorslib](https://github.com/kike-canaries/canairio_sensorlib)

Tinyu-Zhao, 2022-07-15:
You can parameters run this example in your environment https://github.com/m5stack/M5StickC-Plus/blob/master/examples/Basics/AXP192/AXP192.ino

hpsaturn, 2022-07-28:
Thanks for your feedback  and sorry for the delay. Yes with the last version and espressif32 v5.0.0 it works.

closed 2022-07-28 by hpsaturn