m5stack/M5StickC-Plus #55: Wrong IMU pins

sylque, 2024-01-11:
### Describe the bug

The sketch M5StickC-Plus/examples/Basics/IMU/IMU.ino doesn't work because the line `imu6886.Init(32, 33)` is wrong. With my device, it should be `imu6886.Init(GPIO_NUM_21, GPIO_NUM_22)`.

This issue has been difficult to track because your function `imu6886.Init()` returns true even when it fails (in `Wire.begin()`). You might want to fix this also.

### To reproduce

1. Compile and run M5StickC-Plus/examples/Basics/IMU/IMU.ino
=> IMU readings are random

### Expected behavior

IMU readings are correct

### Environment

- OS: Ms Windows
- IDE &IDE Version: VSCode + PlatformIO
- Repository Version: latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5StickC-Plus/issues?q=)
- [X] My report contains all necessary details

sylque, 2024-01-11:
Oups, it seems I'm using the wrong version of IMU.ino... Please let me investigate and clarify this.

sylque, 2024-01-11:
Ok, this is puzzling. There are two sketches for testing the IMU in this repo:
1. https://github.com/m5stack/M5StickC-Plus/tree/master/examples/Basics/IMU
2. https://github.com/m5stack/M5StickC-Plus/tree/master/examples/Unit/IMU_MPU6886

Only the second one has the issue described in this thread, the first one seems ok.

Tinyu-Zhao, 2024-06-24:
You are right, Unit IMU_MPU6886 is an external device, using external I2C.

closed 2024-06-24 by Tinyu-Zhao