m5stack/M5StickC-Plus #56: If (Serial) always returns true

sylque, 2024-01-11:
### Describe the bug

According to [Arduino's documentation](https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/):
> if(Serial)
Description
Indicates if the specified Serial port is ready.
>On the boards with native USB, if (Serial) (or if(SerialUSB) on the Due) indicates whether or not the USB CDC serial connection is open. For all other boards, and the non-USB CDC ports, this will always return true.

The M5StickC-Plus has native USB, so I think `if (Serial)` should return true only when a USB connection is established. This doesn't seem to be the case.

### To reproduce

Use this code from the [Arduino's documentation](https://www.arduino.cc/reference/en/language/functions/communication/serial/ifserial/):
```
void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB
  }

  // Display something in the terminal
  Serial.println("Hello World");
}

void loop() {}
```
This code doesn't display "Hello World" in the terminal, because the line `Serial.println("Hello World");` is called *before* the USB connection is established.

### Expected behavior

You should see "Hello World" in the terminal.

### Environment

- OS: Windows
- IDE &IDE Version: VSCode + PlatformIO
- Repository Version: latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5StickC-Plus/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-06-24:
[image]

Perhaps you just missed that message, establishing a serial connection takes time, and the content in setup will only execute once.

closed 2024-06-24 by Tinyu-Zhao

sylque, 2024-06-24:
@Tinyu-Zhao, I'm not sure I get your point. Sure, you can wait 2 seconds at startup or, even better, use ESP32 log primitives (`log_printf`, etc.). But those workarounds are no fix for people who expect Arduino code to work among platforms.

Tinyu-Zhao, 2024-06-25:
> @Tinyu-Zhao, I'm not sure I get your point. Sure, you can wait 2 seconds at startup or, even better, use ESP32 log primitives (`log_printf`, etc.). But those workarounds are no fix for people who expect Arduino code to work among platforms.

Your device in the establishment of a serial port connection with esp32 takes time, and the serial port initialisation is very fast, successful initialisation that is the implementation of the output, the output of the information is likely to be in the serial port connection is established before you naturally will not see any output from someone!