m5stack/M5StickC-Plus #57: Black display when using setcursor() even when changing fontsizes from 1 to 7

linxcow, 2024-01-27:
### Describe the bug

Using following code after installing m5stickcplus 0.1.0 i get always black display.
I also note in my other projects that using setCursor(0, 0, x) to change font or size  values between 1 and 7 some dont show any display at all , why is that?

          `
          void loop() {
              tftSprite.fillSprite(BLACK);   // Fill the canvas with black.                         
              tftSprite.setCursor(0, 0, 1);  // Set the cursor position and use the 1-point font.
              tftSprite.printf("AXP Temp: %.1fC \r\n",
                               M5.Axp.GetTempInAXP192());  // Get the temperature of AXP192.
                               
              tftSprite.printf("Bat: %.3fv %.1fma\r\n",
                               M5.Axp.GetBatVoltage(),
                               M5.Axp.GetBatCurrent());  // Output voltage and current of // Bat.
                                                         
              tftSprite.printf("Usb: %.2fv %.1fma\r\n",
                               M5.Axp.GetVBusVoltage(),
                               M5.Axp.GetVBusCurrent());  // Output current and voltage of USB
                                                          
              tftSprite.printf("5V-In: %.2fv %.1fma\r\n",
                M5.Axp.GetVinVoltage(), M5.Axp.GetVinCurrent());
              tftSprite.printf("Bat power %.1fmw", M5.Axp.GetBatPower());
              tftSprite.pushSprite(2, 2);   // Where to place our sprite ( 2 pixel marge for best display )
          
              // 0x01 long press(1s), 0x02 press
              // No display or flashing with next line ???
              // if (M5.Axp.GetBtnPress() == 0x02)  { esp_restart(); }  // Restart when powerbutton is pressed shortly
          
              if (M5.BtnA.wasPressed()) {  // When Main button pressed powerdown LCD & Backlight 
                M5.Axp.SetLDO2(false);
              } 
              else if (M5.BtnB.wasPressed()) { // When bottom button pressed powerup LCD & Backlight 
                M5.Axp.SetLDO2(true);
              }
              M5.update();  // Detect whether the keystroke state has changed.
              delay(100);
          }`

### To reproduce

use code above in arduino version above

### Expected behavior

Arduino ide 1.8.19

### Environment

- OS:
- IDE &IDE Version:
- Repository Version:

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5StickC-Plus/issues?q=)
- [X] My report contains all necessary details

linxcow, 2024-01-27:
After a lot of testing I found the the screenbreath() required values have changed nothing is mentioned in the online help
Most of the online help is not uptodate and incomplete. So my code was working but too dim to see

Setting tftSprite.setCursor(0, 0, x);  
font 3 does not display anything also undocumented.

Tinyu-Zhao, 2024-06-24:
https://github.com/m5stack/M5StickC-Plus/assets/44396641/35c4312b-6424-4bb9-92c3-157e2920ecf8

https://github.com/m5stack/M5StickC-Plus/assets/44396641/75621f22-fd4c-4baf-9e68-6c7e68bfd5cb
Are you using the latest library?
I don't reproduce your problem at all, and switching fonts works fine!

Tinyu-Zhao, 2024-07-04:
@linxcow