m5stack/M5StickC-Plus #58: M5StickCPlus example compile error

piaoxuebingfeng, 2024-06-24:
### Describe the bug

 file included from /tmp/.arduinoIDE-unsaved2024524-27760-17nfgmt.nmew/getValue/getValue.ino:2:
/home/qiao/Arduino/libraries/M5StickC-Plus/src/M5StickCPlus.h:119:8: error: extended character “ is not valid in an identifier
  119 | #error “This library only supports boards with ESP32 processor.”
      |        ^
/home/qiao/Arduino/libraries/M5StickC-Plus/src/M5StickCPlus.h:119:64: error: extended character ” is not valid in an identifier
  119 | #error “This library only supports boards with ESP32 processor.”
      |                                                                ^

### To reproduce

in arduino-2.3.2 ,import [M5StickC-Plus](https://github.com/m5stack/M5StickC-Plus)  and compile Button.ino example ,compile error.

### Expected behavior

compile success

### Environment

- OS: deepin-20.9
- IDE &IDE Version: arduino-2.3.2
- Repository Version: [M5StickC-Plus](https://github.com/m5stack/M5StickC-Plus)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/m5stack/M5StickC-Plus/issues?q=)
- [X] My report contains all necessary details

Tinyu-Zhao, 2024-06-24:
Thank you, [fixed](https://github.com/m5stack/M5StickC-Plus/commit/0cc54145dfb939a0fb8d124ff661b2f8717f68b7).

closed 2024-06-24 by Tinyu-Zhao

gaodianzhuo, 2024-08-16:
@Tinyu-Zhao same error,fixed?